vives-project-xp/GreenhouseEnergy #2: Componenten selecteren

SampersS, 2024-09-26:


joshuameuleman, 2024-10-10:
48v stepdown naar 12v 25 A

https://www.amazon.com.be/-/nl/waterdichte-spanningsomvormer-gelijkstroomuitgang-voedingstransformator-elektronische/dp/B09J4WRNZ1/ref=pd_vtp_h_pd_vtp_h_d_sccl_5/260-8109845-6043034?pd_rd_w=iCT5C&content-id=amzn1.sym.d0c92601-62d9-4d90-9dc0-7df2daa2239c&pf_rd_p=d0c92601-62d9-4d90-9dc0-7df2daa2239c&pf_rd_r=EFBE6H257QEWPW8GH7FK&pd_rd_wg=VwymF&pd_rd_r=75a9e77e-acaf-4787-8f2c-3afee45cb331&pd_rd_i=B09J4WRNZ1&psc=1

joshuameuleman, 2024-10-10:
48v stepdown naar 5v 8A

https://www.amazon.com.be/-/nl/waterdichte-spanningsomvormer-gelijkstroomuitgang-voedingstransformator-elektronische/dp/B09J4WRNZ1/ref=pd_vtp_h_pd_vtp_h_d_sccl_5/260-8109845-6043034?pd_rd_w=iCT5C&content-id=amzn1.sym.d0c92601-62d9-4d90-9dc0-7df2daa2239c&pf_rd_p=d0c92601-62d9-4d90-9dc0-7df2daa2239c&pf_rd_r=EFBE6H257QEWPW8GH7FK&pd_rd_wg=VwymF&pd_rd_r=75a9e77e-acaf-4787-8f2c-3afee45cb331&pd_rd_i=B09J4WRNZ1&psc=1

closed 2024-11-04 by OlivierWesterman