vives-project-xp/GreenhouseEnergy #9: Lijst van mogelijke componenten opstellen

OlivierWesterman, 2024-09-26:
Als deel van de research, maak een lijst van opties met daarin mogelijkheden voor batterijen, laadregelaars, vermogensensoren, zonnepanelen. Deze lijst kan later uitegebreid worden indien nodig.

joshuameuleman, 2024-09-26:
vermogensensor uitleg : https://www.eaton.com/us/en-us/products/controls-drives-automation-sensors/sensors---limit-switches/power-sensor.html

vermogenmeter met display
https://www.benselectronics.nl/oled-volt-ampere-vermogens-meter-voor-paneel-inbou.html

SampersS, 2024-10-06:
componenten die we allesinds nodig hebben:
-micro controller: sensor inlezen en communicatie laad eenheid
-temperatuur sensor: per batterij, moet in design passen 
-spanning sensor: per batterij

SampersS, 2024-10-06:
sinds we met redelijk wat cellen werken voorlopig gwn een stm32 gebruiken, wss gaan we dit dan ook nodig hebben in het uiteindelijke resultaat, reden: esp32 heeft wss niet genoeg gpio poorten
Ik neem volgende week wel mijn stm32 mee.

SampersS, 2024-10-06:
<a href='https://www.otronic.nl/nl/ds18b20-to92-temperatuur-sensor-dallas-clone.html'>https://www.otronic.nl/nl/ds18b20-to92-temperatuur-sensor-dallas-clone.html</a>

Dit als temperatuur sensor?

SampersS, 2024-10-06:
> vermogensensor uitleg : https://www.eaton.com/us/en-us/products/controls-drives-automation-sensors/sensors---limit-switches/power-sensor.html
> 
> vermogenmeter met display https://www.benselectronics.nl/oled-volt-ampere-vermogens-meter-voor-paneel-inbou.html

met de nieuwe vereiste van 48V zal dit niet meer gaan (spannings metingen tot 33V)

closed 2024-11-04 by OlivierWesterman