vives-project-xp/GreenhouseEnergy #20: BMS design

SampersS, 2024-10-06:
Meer iets dat sensoren van de batterij uitleest en dan bepaald wat de soc is, ook verstuuren naar de stroombron dat hij moet kappen.
Dit moeten we dan ook toevoegen aan architectuur document.

joshuameuleman, 2024-10-10:
[image]

joshuameuleman, 2024-10-10:
https://www.amazon.com.be/-/nl/13S-48V-wifi-module-communicatiebeschermingskaart-Li-ion/dp/B0CY51CF45/ref=sr_1_5?dib=eyJ2IjoiMSJ9.JvL4Lm1E-8Q6fthPGKmwNhuIVL3FPk5g5Gar2RVyJDyp8RRDLXNdJe4WML4wnp2rV37UmJV4ALrJMCErK9hktq5tFY31sWb3HNB5qf7nttd3SHSEBXYcfSWDks2xHvakJXQBOJqWPaeCw1_ZqisE67xvrXXRd7Mxl48hdifeoVZ-HEwF_VSUTxd8zTzPtD-Pqlyc2IiYbu1Cj4nhdFTzl4p5HMh2o45EvG2Ll5v5HHQ2qwZZdBwXZ8pyocD-FMMQOg__nsan0oxzyJuXMa7QGoABEnCW-biFNuNZugv9zdQ.UORV8f6Y2CpfnhwLHxuDdP-qPcgbc4J-p2OsLC9xyzY&dib_tag=se&qid=1728542297&refinements=p_4%3ADALY&s=electronics&sr=1-5&th=1

joshuameuleman, 2024-10-10:
https://www.dalybms.com/48v-lifepo4-bms-daly-k-series-smart-bms-3s-to-24s-40a-60a-100a-product/?_gl=1*16kdvt6*_up*MQ..*_ga*MTE5ODMzODUyNS4xNzI4NTQzMTI0*_ga_QJJLNQB8EJ*MTcyODU0MzEyMy4xLjEuMTcyODU0MzIwNC4wLjAuMA..*_ga_RPN8G89S7P*MTcyODU0MzEyMy4xLjEuMTcyODU0MzIwNC4wLjAuMTU1NDg5MTUwNg..&gclid=Cj0KCQjw05i4BhDiARIsAB_2wfCFN4HEeqEiV-qE7qk6BNU78S7-xIE7PxOrPT14CxfgQAg40OFrmG0aAotNEALw_wcB

closed 2024-10-10 by OlivierWesterman