vives-project-xp/GreenhouseEnergy #29: 3D ontwerp van modulaire 13s batterij brackets

OlivierWesterman, 2024-10-17:
De brackets moeten 13 batterijen van 18650 model (18mm diameter) naast elkaar op 1 rij kunnen vasthouden.
Afstand tussen de batterijen is 2mm.
Het geheel moet passen op een 3d bed van 25x25 cm, liefst plat voor stevigheid maar kan ook diagonaal geprint worden.

OlivierWesterman, 2024-10-23:
Versie 1.0 die vorige week te klein bleek te zijn aanpassen naar een versie 1.1.
Eenmaal aangepast, nog een proef laten printen om te checken of de nieuwe versie wel groot genoeg is.

Risk: als de versie 1.1 te slecht is afgedrukt, vraag aan mnr.Leroy of je de settings mag specifieren van de 3D printer.

joshuameuleman, 2024-10-24:
versie 1.1 is doorgegeven voor te printen 

wat is er veranderd?

Gaten voor batterij in te steken van 18mm -> 18.4mm
Gaten voor de afstand  tussen top en bottom 3mm -> 3.2mm
Modulaire systeem in het stuk 3.55->3.6   1.55->1.6 
totale hoogte 3,1 maar met 2 delen 1.5 en 1.6 ->1.4 en 1.7

OlivierWesterman, 2024-11-04:
Duplicate, zie #59

closed 2024-11-04 by OlivierWesterman