vives-project-xp/GreenhouseEnergy #38: Ontwerpen brandveiligheid systemen

SampersS, 2024-10-17:
We moeten systemen van verschillende prijs klasses voor brandveiligheid te berijken bedenken en dan voorleggen aan de leerkrachten en daarna kunnen zij kiezen.

OlivierWesterman, 2024-10-23:
Kijk hiervoor in de map Design voor een file met de naam Brandveiligheid_voorstellen.

Wat er concreet moet gebeuren:
Geef een drietal verschillende ideeen voor hoe we de batterij brandveilig zouden kunnen plaatsen.
Laat die varieren in prijs en kwaliteit, dus eentje goedkoop maar minder veilig, een middenslag en een duur maar betrouwbaar systeem.

Geef daar dus ook een geschatte kostprijs voor ieder concept en een ruw idee van welk materiaal je zou gebruiken bij elk.

Stuur gerust via teams als je nog een vraag hierover hebt.

joshuameuleman, 2024-10-24:
ik zal hier mogelijke sites zetten

https://www.emtez.be/nl-be/lithium-ion-quarantine-safety-cabinets : een aan te kopen kast die ontworpen is precies om lithium-ion batterijen op te bergen: heeft ook al een quarantaine bak bij 70°C zal de planken waar de batterij opstaat omkieperen

OlivierWesterman, 2024-11-04:
afgewerkt, zie Design>Brandveiligheid_voorstellen en #60

closed 2024-11-04 by OlivierWesterman