vives-project-xp/GreenhouseEnergy #42: Correcte fuses vinden online

OlivierWesterman, 2024-10-23:
We moeten nog fuses bestellen maar die zijn niet zo eenvoudig te vinden. fuse = zekering
Specifiek gaat het over:
1 fuse die op 48V 30Amp triggert
1 fuse die op 48V 20Amp triggert

Zoek een website die deze verkoopt en post de links in deze issue.

joshuameuleman, 2024-10-24:
Dit is een auto zekering die in uw zekeringsdoos zit van uw auto

deze zekering/fuse kan 30A op 48v aan. 

voordeel: het heeft onze specs dat we nodig hebben

nadeel: nood aan een zekeringsdoos 

https://www.digikey.be/nl/products/detail/optifuse/APT-30A/11581699

dit is van amazon kan 48v aan en je kan kopen in 20A en 30A 
https://www.amazon.com/Hamolar-Circuit-Breaker-Resettable-Inverter/dp/B0B51X8FJ3?th=1

OlivierWesterman, 2024-11-04:
Toegevoegd aan bestellijst Excel.

closed 2024-11-04 by OlivierWesterman