vives-project-xp/GreenhouseEnergy #82: Onderzoek data verkrijgen van Victron MPPT charger

OlivierWesterman, 2024-11-07:
Onze project owner heeft ons gevraagd om al vooruit te denken over hoe we eventueel data zouden kunnen verkrijgen van de MPPT charger, sinds die er duidelijk verzameld en via een propietary app via bluetooth doorstuurt.

Zoek online op of er al andere mensen hier iets op gevonden hebben en/of Victron dit ondersteund.
Zet hierbij alles dat je vindt in de folder Research.

OlivierWesterman, 2024-11-21:
VE direct protocol, met UART aansluiten te vinden op de charger zelf. 
Via een microcontroller zoals een ESP32 kunnen we de data ontvangen en doorsturen via Wifi/Bluetooth.

SampersS, 2024-11-30:
Na het lezen van de documentatie over VE direct is vastgesteld dat we dit kunnen toepassen op een esp32 en dat het onderzoek dus voltooid is.

closed 2024-11-30 by SampersS